STMicroelectronics/stm32f7xx-hal-driver #2: Bug in stm32f7xx_hal_hcd.c using callback

micheledarold, 2021-06-24:
Missing use of registered callback in some part of the code.

I have enabled USE_HAL_HCD_REGISTER_CALLBACKS but I find some part of the code not using the correct way to call the function
HAL_HCD_HC_NotifyURBChange_Callback.
[hcd_patch.txt](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/files/6707072/hcd_patch.txt)

ALABSTM, 2021-08-09:
Hi @micheledarold,

Thank you for this report. Your point is correct. It will be forwarded to our development teams. May I ask you first to create a pull-request? This will allow a far more easy way to view the changes proposed. Thank you very much.

With regards,

ALABSTM, 2021-08-11:
ST Internal Reference: 111775

ALABSTM, 2021-08-11:
Hi @micheledarold,

The issue you reported has been forwarded to our development teams. No more need to create a pull-request this time. Next times maybe, as it allows to visualize the changes far more easily and, more important, to integrate them faster into the repository. Thank you again for having reported this one.

With regards,

micheledarold, 2021-08-23:
Hi we was on vacations, I'll provide the pull today or tomorrow

With regards

micheledarold, 2021-08-23:
[0001-issue-2.zip](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/files/7030070/0001-issue-2.zip)

Hi @ALABSTM is ok for you the patch file?

micheledarold, 2021-08-23:
Hi @ALABSTM I'm on vacation and bak in office only today, so thank you for your help, next time I'll provide the pull request.

ALABSTM, 2021-12-15:
Hi @micheledarold,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.2.10** of the stm32f7xx_hal_driver published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-15 by ALABSTM