STMicroelectronics/stm32f7xx-hal-driver #10: HAL_PCD_SetTestMode() doesn't clear register bits before setting them

kevinpt, 2024-07-10:
**Describe the set-up**
 * STM32F723
 * IDE 1.14.1, HAL 1.3.1

**Describe the bug**
Inspection of the code for [HAL_PCD_SetTestMode()](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/blob/2e3aac48671f6f5d86c8e85b314b50589c1dc93c/Src/stm32f7xx_hal_pcd.c#L2031) shows that the RW bits for TCTL in the OTG_DCTL register are not cleared before setting them. This function will work on the first usage but future changes to test mode will garble the bits.

For completeness there should probably also be a define for test mode disabled (0U).

ALABSTM, 2024-07-12:
ST Internal Reference: 186507

ALABSTM, 2025-07-28:
Fixed in release **1.3.2**