STMicroelectronics/stm32f7xx-hal-driver #13: Cannot program internal flash with double word

satur9nine, 2025-03-25:
**Describe the set-up**
 * NUCLEO-F767ZI
 * ARM NONE EABI GCC 13.3

**Describe the bug**
Flash program PGP error when trying to write double word to address 0x08010000

**How To Reproduce**
Use the following code:

    HAL_StatusTypeDef status = HAL_OK;

    HAL_FLASH_Unlock();

    uint64_t double_word_data 0xAA;
    status = HAL_FLASH_Program(FLASH_TYPEPROGRAM_DOUBLEWORD, 0x08010000, double_word_data);

    HAL_FLASH_Lock();

When using just single WORD programming it works. Datasheet DS11532 Rev 8 for "STM32F765xx STM32F767xx STM32F768Ax STM32F769xx" never mentions parallelism x64 anywhere. I think this chip doesn't really support double word programming. There are no examples of double word programming in STM32F7Cube. If chip doesn't support it then driver should not offer it, define FLASH_TYPEPROGRAM_DOUBLEWORD should be removed or more information provided how to make it work or HAL fixed so it just works.

KRASTM, 2025-03-28:
Hello @satur9nine,

Thank you for the report.

In fact, the DS11532 Rev 8 datasheet contains a table **Table 58. Flash Memory Programming with VPP** on page 152 for double word programming, also in the driver HAL file and in the short section for **static void FLASH_Program_DoubleWord()**, there is a note ` @note   This function must be used when the device voltage range is from 2.7V to 3.6V and an External Vpp is present.`. I suggest you take a look and try it out with this configuration.

In the meantime, I will check with our team for more information and keep you posted.

With regards,

satur9nine, 2025-03-31:
Thanks since I am not calling the function `FLASH_Program_DoubleWord` directly I did not read the source code documentation on it. Would be ideal if the `FLASH_TYPEPROGRAM_DOUBLEWORD` parameter has the same code documentation mentioned that Vpp is required to use this parameter.

KRASTM, 2025-04-08:
Hello @satur9nine,

I think so, I will share your opinion with our team. Please allow me to close the issue.

Thank you for your understanding.

Best Regards,

closed 2025-04-08 by KRASTM