STMicroelectronics/stm32f7xx-hal-driver #14: 8-Bit NOR devices cause hard fault on Hal_Nor_Init

StephenPPEng, 2025-04-08:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version
 Custom STM32F756BG Board.
Macronix MX29LV040CTI (or any other 8-bit NOR flash)
Visual Studio IDE; Code generated from STM32CubeMx.

**Describe the bug**
in HAL_NOR_Init, the code Hard Faults when setting the Nor ->CommandSet.
Additionally, before this, the NOR_WRITE is writing 16-bit values to a 8-bit device.

**How To Reproduce**
1- Spin up a project in stm32cubemx that has a NOR flash on NORSRAM_Bank_3 in Write mode (8-bit data)
2-Ensure the MPU is configured correctly
3- Generate Code
4- Run the code on a compatible board/device

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
HAL -> NOR

3. The use case that generates the problem
When running HAL_NOR_Init in the FMC for a 8-bit parallel NOR flash chip
4. How we can reproduce the problem
**Additional context**

ISSUE1:
NOR_WRITE casts addresses to a 16-bit pointer.  So when 'writing' to a 8-bit Flash, it is writing 2 bytes instead of just 1.
ISSUE2:
`hnor->CommandSet = *(__IO uint16_t *) NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_ADDRESS_COMMAND_SET);`
This line also has an explicit 16-bit cast.  WHen working with 8-bit flash, this should be an 8-bit cast - THe  16-bit cast is causing a mis-aligned memory access when reading the command set (the address is 0x13... which is odd).
Odd-alligned memory access is prohibited for 16-bit pointers.
ISSUE3:
this section
```
    /* Get the value of the command set */
    if (uwNORMemoryDataWidth == NOR_MEMORY_8B)
    {
      NOR_WRITE(NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST_CFI_BYTE),
                NOR_CMD_DATA_CFI);
    }
    else
    {
      NOR_WRITE(NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST_CFI), NOR_CMD_DATA_CFI);
    }
```
is meaningless as written.  the NOR_ADDR_SHIFT already takes the uwNORMemoryDataWidth into account.  There does not need to be 2 separate calls for 2 addresses since the NOR_ADDR_SHIFT already modifies the address.

**Screenshots**
If applicable, add screenshots to help explain your problem.

StephenPPEng, 2025-04-08:
My suggested fix;  Or at least, something that works.

in stm32f7xx_hal_nor.c
[image]

in stm32f7xx_hal_nor.h
[image]

StephenPPEng, 2025-04-08:
After evaluating some more, i do not see a straightforward solution to this problem.
The easiest workaround, for now, is to declare the NOR flash as a 16-bit data width, and everything works flawlessly.... except every other byte is 0xFF!

StephenPPEng, 2025-04-09:
Spent the last day looking at this;
This is all some really strange behavior...
1. My NOR flash chip uses FUJITSU command set, and is an 8-bit data bus.  However, for the CFI Read, it requires writing `0x98` to _**0x55**_ instead of _**0xAA**_, as seen in my code snippet above.  From my understanding, this is strange behavior.
2. There are many spots in the HAL where `NOR_WRITE` calls must be cast to a `__IO uint8_t *` to prevent memory misallignment when trying to use the default `__IO uint16_t *`; these change must be made in the supplied HAL itself.
3. When i identify my NOR chip as a 16-bit data bus, the `commandSet` reads a `0xFF02`, which is obviously not the expected `0x02` of the Fujitsu command set..... however sometimes my code still runs somehow.

I'm currently in the process of modifying the HAL to *correctly* support both 8-bit and 16-bit data buses for my purposes.  Let me know if you need any info.

StephenPPEng, 2025-04-09:
> My suggested fix; Or at least, something that works.
> 
> in stm32f7xx_hal_nor.c [image]
> 
> in stm32f7xx_hal_nor.h [image]

[image]
A cleaner version of what i had written earlier.

KRASTM, 2025-04-11:
Hello @StephenPPEng,

Thank you for the report.

Regarding the issue, and after a first check in the **RM0385**, we have a chapter `13.5 External device address mapping` at the page 312 describing how the external memory is divided into fixed-size banks of 256 Mbytes each, below an extract of the different banks and the supported memory type for each one (for the bank 3, we have only NAND flash memory):

[image]

So, if you may share with us, your ioc File and the version of CubeMX you are using, we can analyze it further.

With regards,

StephenPPEng, 2025-04-11:
Absolutely, once I am back at my computer I will share the .ioc
For now, I should clarify:
I have my NOR on Bank 1(NOR/SRAM), Sub bank 3 (NE3).
MPU is modified to allow full access to 0x68000000+512k.
The NOR chip is byte-addressed (8 bit) only, but supports CFI in 16-Bit mode.

Best,
Stephen

image.png (view on web)<https://github.com/user-attachments/assets/098dbff1-cb3b-42c7-8938-3970689c1cc4>

image.png (view on web)<https://github.com/user-attachments/assets/098dbff1-cb3b-42c7-8938-3970689c1cc4>

This email is intended solely for the use of the individual to whom it is addressed and may contain confidential and/or privileged material. Any views or opinions presented are solely those of the author and do not necessarily represent those of Precision Planting. If you are not the intended recipient, be advised that you have received this email in error and that any use, dissemination, forwarding, printing, or copying of this email is strictly prohibited. Neither AGCO nor the sender accepts any responsibility for viruses, and it is your responsibility to scan, and virus check the e-mail and its attachment(s) (if any).

StephenPPEng, 2025-04-14:
@KRASTM 
CubeMx, 6.14.0

```
#MicroXplorer Configuration settings - do not modify
CAD.formats=[]
CAD.pinconfig=Dual
CAD.provider=
CORTEX_M7.AccessPermission_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=MPU_REGION_FULL_ACCESS
CORTEX_M7.AccessPermission_S-Cortex_Memory_Protection_Unit_Region2_Settings_S=MPU_REGION_FULL_ACCESS
CORTEX_M7.AccessPermission_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=MPU_REGION_FULL_ACCESS
CORTEX_M7.AccessPermission_S-Cortex_Memory_Protection_Unit_Region4_Settings_S=MPU_REGION_FULL_ACCESS
CORTEX_M7.BaseAddress_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=0xA0000000
CORTEX_M7.BaseAddress_S-Cortex_Memory_Protection_Unit_Region2_Settings_S=0x60000000
CORTEX_M7.BaseAddress_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=0x80000000
CORTEX_M7.BaseAddress_S-Cortex_Memory_Protection_Unit_Region4_Settings_S=0x68000000
CORTEX_M7.DisableExec_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=MPU_INSTRUCTION_ACCESS_DISABLE
CORTEX_M7.DisableExec_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=MPU_INSTRUCTION_ACCESS_DISABLE
CORTEX_M7.Enable_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=MPU_REGION_ENABLE
CORTEX_M7.Enable_S-Cortex_Memory_Protection_Unit_Region2_Settings_S=MPU_REGION_ENABLE
CORTEX_M7.Enable_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=MPU_REGION_ENABLE
CORTEX_M7.Enable_S-Cortex_Memory_Protection_Unit_Region4_Settings_S=MPU_REGION_ENABLE
CORTEX_M7.Enable_Spec=MPU_REGION_ENABLE
CORTEX_M7.IPParameters=default_mode_Activation,Enable_Spec,Enable_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,BaseAddress_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,Size_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,AccessPermission_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,DisableExec_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,IsShareable_S-Cortex_Memory_Protection_Unit_Region1_Settings_S,Enable_S-Cortex_Memory_Protection_Unit_Region2_Settings_S,BaseAddress_S-Cortex_Memory_Protection_Unit_Region2_Settings_S,Size_S-Cortex_Memory_Protection_Unit_Region2_Settings_S,AccessPermission_S-Cortex_Memory_Protection_Unit_Region2_Settings_S,IsShareable_S-Cortex_Memory_Protection_Unit_Region2_Settings_S,Enable_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,BaseAddress_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,Size_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,AccessPermission_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,DisableExec_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,IsShareable_S-Cortex_Memory_Protection_Unit_Region3_Settings_S,Enable_S-Cortex_Memory_Protection_Unit_Region4_Settings_S,BaseAddress_S-Cortex_Memory_Protection_Unit_Region4_Settings_S,Size_S-Cortex_Memory_Protection_Unit_Region4_Settings_S,IsShareable_S-Cortex_Memory_Protection_Unit_Region4_Settings_S,AccessPermission_S-Cortex_Memory_Protection_Unit_Region4_Settings_S
CORTEX_M7.IsShareable_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=MPU_ACCESS_SHAREABLE
CORTEX_M7.IsShareable_S-Cortex_Memory_Protection_Unit_Region2_Settings_S=MPU_ACCESS_SHAREABLE
CORTEX_M7.IsShareable_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=MPU_ACCESS_SHAREABLE
CORTEX_M7.IsShareable_S-Cortex_Memory_Protection_Unit_Region4_Settings_S=MPU_ACCESS_SHAREABLE
CORTEX_M7.Size_S-Cortex_Memory_Protection_Unit_Region1_Settings_S=MPU_REGION_SIZE_8KB
CORTEX_M7.Size_S-Cortex_Memory_Protection_Unit_Region2_Settings_S=MPU_REGION_SIZE_1MB
CORTEX_M7.Size_S-Cortex_Memory_Protection_Unit_Region3_Settings_S=MPU_REGION_SIZE_1GB
CORTEX_M7.Size_S-Cortex_Memory_Protection_Unit_Region4_Settings_S=MPU_REGION_SIZE_1MB
CORTEX_M7.default_mode_Activation=1
Dma.I2C2_TX.0.Direction=DMA_MEMORY_TO_PERIPH
Dma.I2C2_TX.0.FIFOMode=DMA_FIFOMODE_DISABLE
Dma.I2C2_TX.0.Instance=DMA1_Stream7
Dma.I2C2_TX.0.MemDataAlignment=DMA_MDATAALIGN_BYTE
Dma.I2C2_TX.0.MemInc=DMA_MINC_ENABLE
Dma.I2C2_TX.0.Mode=DMA_NORMAL
Dma.I2C2_TX.0.PeriphDataAlignment=DMA_PDATAALIGN_BYTE
Dma.I2C2_TX.0.PeriphInc=DMA_PINC_DISABLE
Dma.I2C2_TX.0.Priority=DMA_PRIORITY_LOW
Dma.I2C2_TX.0.RequestParameters=Instance,Direction,PeriphInc,MemInc,PeriphDataAlignment,MemDataAlignment,Mode,Priority,FIFOMode
Dma.Request0=I2C2_TX
Dma.RequestsNb=1
FATFS.IPParameters=_USE_LFN,_FS_EXFAT
FATFS._FS_EXFAT=1
FATFS._USE_LFN=1
FMC.AddressSetupTime3=10
FMC.AttributeSpaceHiZSetupTime1=4
FMC.AttributeSpaceHoldSetupTime1=3
FMC.AttributeSpaceSetupTime1=3
FMC.AttributeSpaceWaitSetupTime1=5
FMC.BusTurnAroundDuration1=6
FMC.BusTurnAroundDuration3=5
FMC.CommonSpaceHiZSetupTime1=4
FMC.CommonSpaceHoldSetupTime1=3
FMC.CommonSpaceSetupTime1=3
FMC.CommonSpaceWaitSetupTime1=5
FMC.DataSetupTime1=8
FMC.DataSetupTime3=20
FMC.ECCPageSize1=FMC_NAND_ECC_PAGE_SIZE_512BYTE
FMC.EccComputation1=FMC_NAND_ECC_ENABLE
FMC.IPParameters=NandPageSize1,NandSpareAreaSize1,NandBlockSize1,NandBlockNbr1,NandPlaneNbr1,NandPlaneSize1,EccComputation1,ECCPageSize1,AttributeSpaceHiZSetupTime1,AttributeSpaceSetupTime1,AttributeSpaceWaitSetupTime1,AttributeSpaceHoldSetupTime1,CommonSpaceSetupTime1,CommonSpaceWaitSetupTime1,CommonSpaceHoldSetupTime1,CommonSpaceHiZSetupTime1,TCLRSetupTime1,TARSetupTime1,DataSetupTime1,BusTurnAroundDuration1,WriteOperation1,WriteOperation3,AddressSetupTime3,DataSetupTime3,BusTurnAroundDuration3
FMC.NandBlockNbr1=8192
FMC.NandBlockSize1=64
FMC.NandPageSize1=2048
FMC.NandPlaneNbr1=2
FMC.NandPlaneSize1=4096
FMC.NandSpareAreaSize1=64
[... 494 lines cut ...]
SH.FMC_NWE.ConfNb=3
TIM3.IPParameters=Period,Prescaler
TIM3.Period=5000
TIM3.Prescaler=10799
[... cut ...]
VP_SYS_VS_Systick.Mode=SysTick
VP_SYS_VS_Systick.Signal=SYS_VS_Systick
VP_TIM3_VS_ClockSourceINT.Mode=Internal
VP_TIM3_VS_ClockSourceINT.Signal=TIM3_VS_ClockSourceINT
VP_USB_HOST_VS_USB_HOST_MSC_FS.Mode=MSC_FS
VP_USB_HOST_VS_USB_HOST_MSC_FS.Signal=USB_HOST_VS_USB_HOST_MSC_FS
board=custom
```
Additionally, here is the final version of how to get the Command Set (notice the 8-bit NOR_WRITE).  This can always be 8-bit because the standard incicates that the return value for the command set will be 1 byte.

[image]

StephenPPEng, 2025-04-14:
So what I've found in my testing:

The NOR_WRITE() function needs to match the size of the Data Width.
If Data Width is 8-bits, then the `NOR_WRITE()` needs to cast to a `(__IO uint8_t *)`.
If 16-bits, then the `NOR_WRITE()` needs to cast to a `(__IO uint16_t *)`.
etc.

The NOR_ADDR_SHIFT already does the job of moving the correct value to the right spot.  There does not need to be "special" 8-bit vs 16-bit command bytes.

Generated Code for HAL_NOR_Init():  
[image]
Issues are circled.

Green:  This should be an 8-bit `NOR_WRITE()` instead of a 16-bit `NOR_WRITE()`. Since 8-bit addresses can be odd, using a 16-bit `NOR_WRITE()` here could/will cause an Unalligned Memory Access fault.

Red: This should be `NOR_CMD_ADDRESS_FIRST_CFI`.  It SHOULD NOT be `NOR_CMD_ADDRESS_FIRST_CFI_BYTE` because the NOR_ADDR_SHIFT handles shifting the address to the correct bits. (The correct address to Write to for this command is `0x55`, not `0xAA`, according to the CFI Spec Document)

Blue:  Any time values are READ from NOR, the address to read should be casted to the appropriate sized pointer.  Here, it is always cast as a 16-bit pointer.  This is a problem.  Another option is to ALWAYS use an 8-bit pointer.  Well now you are only getting 1 byte on a 16-bit data bus.

| Device Type / Mode | Result |
| --- | --- |
| x8 Device / x8 Read / Odd Address | No fault. Works as expected.  Read will retrieve 1 byte from NOR and put 1 byte into the buffer. |
| x8 Device / x8 Read / Even Address | No fault. Works as expected  Read will retrieve 1 byte from NOR and put 1 byte into the buffer. |
| x8 Device / x16 Read / Odd Address | Misalligned Memory Access Fault |
| x8 Device / x16 Read / Even Address | No fault. Value must be masked off with `val & 0xFF` to ensure upper trash byte is not forwarded |
| x16 Device / x8 Read / Odd Address | No Fault. But you are sending an odd address to a 16-bit device, so behavior may be device-dependent.  My device simply removed the LSb from the address to make it even and then returned that value.  But the buffer will only be filled with 1 byte, instead of the expected 2 bytes.
| x16 Device / x8 Read / Even Address | No Fault.  But the buffer will only be filled with 1 byte instead of the expected 2 bytes.
| x16 Device / x16 Read / Odd Address | No Fault.  Works as expected.  Read will retrieve 2 bytes from NOR and put 2 bytes into the buffer.
| x16 Device / x16 Read / Even Address | No Fault. Works as expected.  Read will retrieve 2 bytes from NOR and put 2 bytes into the buffer.

StephenPPEng, 2025-04-14:
TL;DR:
Original Generated Code for the end of the `HAL_NOR_Init()`:
```
...
else
  {
    /* Get the value of the command set */
    if (uwNORMemoryDataWidth == NOR_MEMORY_8B)
    {
      NOR_WRITE(NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST_CFI_BYTE),
                NOR_CMD_DATA_CFI);
    }
    else
    {
      NOR_WRITE(NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST_CFI), NOR_CMD_DATA_CFI);
    }

    hnor->CommandSet = *(__IO uint16_t *) NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_ADDRESS_COMMAND_SET);

    status = HAL_NOR_ReturnToReadMode(hnor);
  }
...
```
My suggested change:
```
...
else
  {
    /* Get the value of the command set */
    NOR_WRITE_8b(NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST_CFI), 
                NOR_CMD_DATA_CFI);
    hnor->CommandSet = *(__IO uint8_t *)NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_ADDRESS_COMMAND_SET);

    status = HAL_NOR_ReturnToReadMode(hnor);
  }
...
/**
  * @brief  NOR memory write 8-bit data to specified address.
  * @param  __ADDRESS__ NOR memory address
  * @param  __DATA__ Data to write
  * @retval None
  */
#define NOR_WRITE_8b(__ADDRESS__, __DATA__)   do{                                                             \
                                               (*(__IO uint8_t *)((uint32_t)(__ADDRESS__)) = (__DATA__)); \
                                               __DSB();                                                    \
                                             } while(0)
```

And this needs to be done everywhere in the HAL_NOR file.
To support different data types, HAL_NOR_Read() and HAL_NOR_Program() must have their signatures changed to the following

`HAL_StatusTypeDef HAL_NOR_Read(NOR_HandleTypeDef *hnor, uint32_t *pAddress, uint16_t *pData); `
becomes
`HAL_StatusTypeDef HAL_NOR_Read(NOR_HandleTypeDef *hnor, uint8_t *pAddress, void *pData);`

and
`HAL_StatusTypeDef HAL_NOR_Program(NOR_HandleTypeDef *hnor, uint32_t *pAddress, uint16_t *pData);`
becomes
`HAL_StatusTypeDef HAL_NOR_Program(NOR_HandleTypeDef *hnor, uint8_t *pAddress, void *pData);`

KRASTM, 2025-04-15:
ST Internal Reference: 207766

StephenPPEng, 2025-04-15:
For another reference, here is what my HAL_NOR_Program() looks like
```C
/**
  * @brief  Program data to NOR memory
  * @param  hnor pointer to a NOR_HandleTypeDef structure that contains
  *                the configuration information for NOR module.
  * @param  pAddress Device address
  * @param  data the data to be programmed
  * @retval HAL status
  */
HAL_StatusTypeDef HAL_NOR_Program(NOR_HandleTypeDef *hnor, uint8_t *pAddress, uint32_t data)
{
  uint32_t deviceaddress;
  HAL_StatusTypeDef status = HAL_OK;

  /* Check the NOR controller state */
  if (hnor->State == HAL_NOR_STATE_BUSY)
  {
    return HAL_BUSY;
  }
  else if (hnor->State == HAL_NOR_STATE_READY)
  {
    /* Process Locked */
    __HAL_LOCK(hnor);

    /* Update the NOR controller state */
    hnor->State = HAL_NOR_STATE_BUSY;

    /* Select the NOR device address */
    if (hnor->Init.NSBank == FMC_NORSRAM_BANK1)
    {
      deviceaddress = NOR_MEMORY_ADRESS1;
    }
    else if (hnor->Init.NSBank == FMC_NORSRAM_BANK2)
    {
      deviceaddress = NOR_MEMORY_ADRESS2;
    }
    else if (hnor->Init.NSBank == FMC_NORSRAM_BANK3)
    {
      deviceaddress = NOR_MEMORY_ADRESS3;
    }
    else /* FMC_NORSRAM_BANK4 */
    {
      deviceaddress = NOR_MEMORY_ADRESS4;
    }

    /* Send program data command */
    if (hnor->CommandSet == NOR_AMD_FUJITSU_COMMAND_SET)
    {
      uint32_t cycle1Addr = NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_FIRST);
      uint32_t cycle2Addr = NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_SECOND);
      uint32_t cycle3Addr = NOR_ADDR_SHIFT(deviceaddress, uwNORMemoryDataWidth, NOR_CMD_ADDRESS_THIRD);
      if (uwNORMemoryDataWidth == NOR_MEMORY_8B)
      {
        NOR_WRITE_8b(cycle1Addr, NOR_CMD_DATA_FIRST);
        NOR_WRITE_8b(cycle2Addr, NOR_CMD_DATA_SECOND);
        NOR_WRITE_8b(cycle3Addr, NOR_CMD_DATA_PROGRAM);
      }
      else
      {
        NOR_WRITE(cycle1Addr, NOR_CMD_DATA_FIRST);
        NOR_WRITE(cycle2Addr, NOR_CMD_DATA_SECOND);
        NOR_WRITE(cycle3Addr, NOR_CMD_DATA_PROGRAM);
      }
    }
    else if (hnor->CommandSet == NOR_INTEL_SHARP_EXT_COMMAND_SET)
    {
      NOR_WRITE(pAddress, NOR_CMD_WORD_PROGRAM);
    }
    else
    {
      /* Primary command set not supported by the driver */
      status = HAL_ERROR;
    }

    if (status != HAL_ERROR)
    {
      /* Write the data */
      if(uwNORMemoryDataWidth == NOR_MEMORY_8B)
      {
        NOR_WRITE_8b(pAddress, data & 0xFF);
      }
      else
      {
        pAddress = (uint8_t *)((uint32_t)pAddress & 0xFFFFFFFE); /* Address must be even for 16 bits access */
        NOR_WRITE(pAddress, data & 0xFFFF);
      }
    }
    
    /* Check the NOR controller state */
    hnor->State = HAL_NOR_STATE_READY;

    /* Process unlocked */
    __HAL_UNLOCK(hnor);
  }
  else
  {
    return HAL_ERROR;
  }

  return status;
}

```

closed 2026-02-13 by KRASTM